fossasia/pslab-python #13: Merge repositories to be more user friendly

mariobehling, 2016-07-13:
The readme.md already states "clone two repositories". Why we have two repositories then? Please merge the repos and import "apps" as a folder into this repo.

wavicles, 2016-07-14:
@mariobehling  Two separate repositories are created because,  if users are only interested in using PSLab as an acquisition device without a display, the don't need to download the huge graphics dependencies such as pyqt4, pyqtgraph and screenshots etc.

wavicles, 2016-07-14:
Some of the Debian package maintainers follow this method. But If required it is possible to merge  pslab-apps repo with pslab.

mariobehling, 2016-07-14:
@wavicles People can still use PSLab as an acquisition device and the code stays separate. Please just add it here into a folder. Debian people can still follow their procedure as well.

mariobehling, 2016-07-14:
@wavicles Actually this also depends on the question if there might be a chance to implement the whole thing as a web app that runs in the browser.

wavicles, 2016-07-15:
Yes. This is possible.  I will start working on this idea.

wavicles, 2016-07-21:
@mariobehling  For implementing the whole thing as a web app, some sort of server needs to be present for hosting the files, right?

wavicles, 2016-07-21:
If we want PSLab to be used as stand alone tool, then the server and wifi needs to run on PSLab itself. In which case, an ESP8266 module can be added to PSLab board and server code must be uploaded on to it.

mariobehling, 2016-07-21:
Well, ok, it is possible to build a device, but then the goal is not anymore to run it on any PC. So, this would mean PSLabs comes on a Raspi or similar with a wifi chip. Then it cannot be simply installed on a PC. So, what really needs to be defined is the use case and ask ourselves: What do people have available to get them started? PC, mico-computers like Raspis, mobile phone? Who do you want to enable? If you want to have the broadest possible outreach, it should probably be a phone app and it could be a web app running on the phone as well. We could have all kinds of syncs with Firebase or other technologies. So, what is your ultimate goal?

closed 2017-04-07 by mariobehling