fossasia/pslab-python #15: Implementation with browser web UI options

mariobehling, 2016-07-13:
What are the options to implement this project and make it accessible through the browser? This would have huge advantages in regards to the devices that can be used.

wavicles, 2016-07-15:
@mariobehling  Yes. It is possible to make PSLab accessible through the browser . We will soon start working on this idea. 
For web app Django will be a better option. It can also be done with existing framework, wherein one needs to use HTML and JQUERY. I think Django might be more efficient option.

wavicles, 2016-07-15:
@mariobehling  At present I am working on remote access. This can be done in two/three different options
A) CherryPy based web server
this is a REST framework, where commands can be executed using a web browser
e.g. 192.168.1.1:8888/get_voltage('CH1') , where a machine with IP address 192.168.1.1 must be running PSLab , and be connected on the network
B) PubNub
Pubnub is a service for IoT devices. It is something like Hangouts for devices

Jaminy, 2016-11-01:
Hi, 
I would like to work on this and I hope it would  be helpful for everyone to get accessed.

souravsingh, 2017-09-14:
I believe this fixed through this- https://github.com/fossasia/pslab-remote

CloudyPadmal, 2019-05-14:
Electron app is being built and it will address this issue.

closed 2019-05-14 by CloudyPadmal