fossasia/pslab-python #35: Develop an Android App for PSLab Experiments

wavicles, 2017-03-15:
There are a number of experiments currently running with PSLab. These are written as desktop experiments in Python. The next phase of the project enables users to make use of their mobile phone. Please propose an architecture, that does not necessarily require a server and offers teachers and students a way to make full use of the PSLab architecture.

PSLab is a new addition to FOSSASIA Science Lab. This tiny pocket lab provides an array of necessary equipments for doing science and engineering experiments. It can function like an oscilloscope, waveform generator, frequency counter, programmable voltage and current source and also as a data logger. The control and measurement functions are written in Python programming language. Pyqtgraph is used for plotting library. A variety of experiments can be performed using GUI applications written in python. PSLab is interfaced and powered by USB port of the computer.

Repository: https://github.com/fossasia/pslab

Chat: https://fossasia.slack.com/messages/pocketscience/details/ | [Get an invite]

Website: http://pslab.fossasia.org

https://github.com/fossasia/labs.fossasia.org/issues/206

mohitmanuja, 2017-03-15:
I am trying to understand this project, can you provide me with some pointers from where I can understand how input is received in mobile.

jithinbp, 2017-03-15:
Hi,

There are two approaches to this, given the hardware architecture currently
being used.
A) port the python communication library as well as apps to Java . This
will enable running PSLab on OTG enabled android phones (most modern phones
have OTG support)
The device needs to be physically connected to the phone via a cable.

B) utilise the provision in the hardware to add an ESP8266 WiFi module for
communication . The current footprint is for an ESP12E module.

- The ESP module can run a lightweight web server hosting CGI callbacks for
reading data from the device, and Bootstrap based front-end apps for
experiments can be made. esphttpd web server can be an open source starting
point [written by Jeroen ( sprite_tm )].
In this configuration, the device needs to be externally powered by a 5V
supply, but can be accessed wirelessly by any device with a web browser. No
native apps required. ESP has both AP & Station modes.

- load the ESP module with a transparent TCP-UART bridge ( such as the one
by JEELabs ) , and design native apps that communicate via TCP.

A decision needs to be made regarding the direction to be taken. I believe
the standalone web server approach opens up a wider user base, and will
reduce the coding effort for multiple platforms.
I've written partial / related work in both directions. Let me know what
will be preferable .

<https://github.com/fossasia/pslab/issues/35#issuecomment-286813064>, or mute
the thread

viveksb007, 2017-03-15:
@jithinbp I agree with the standalone web-server approach as it prevents writing communication interface for every platform it needs to be compatible with.
So we need to write CGI callbacks that server on ESP8266 would host. This eliminates the need for an Android App as we can access it directly through browser. We can use it with any platform like Android, iOS, Windows. But for someone who doesn't have experience with embedded systems, it would be difficult to connect and host server on esp8266 module.
On the other hand, android device is easily accessible to anyone, so we can safely go with approach A too.
@wavicles @mariobehling  So finally what should we start working for?

akarshan96, 2017-03-16:
I think using ESP module for communication between the Android Smartphone and PSLab is better option @jithinbp. I agree @viveksb007, this will allow PSLab to communicate with any platform hence widens the future scope and will support further developments.

mohitmanuja, 2017-03-16:
@jithinbp 
Going with approach 'A' will not be difficult but it will be a replicating the same task which already has been done, we might need to do this for other platforms, we can put this energy to make pslab better and also, it will require synchronization the way things are getting done on each type of platform, so it should be the least priority if aren't left with any other options.

mohitmanuja, 2017-03-19:
@wavicles @jithinbp Hey, Please read my above reply and help to get started.

jithinbp, 2017-04-08:
I think we should stick with the native android app option, as we already have developers who seem up to the task. Open to suggestions.

@mohitmanuja I believe the ESP option also ties us down to the capabilities of the ESP. The user will need to power the device through a separate adapter, as the ESP consumes up to 300mA. The device otherwise consumes <130mA .   We have already covered the PC platform, and Android is dominant in the mobile market, so I think ESP won't give us any significant edge despite requiring  more development effort (ESP comm , CGI callbacks, JSON packing , + Frontend).

jithinbp, 2017-06-08:
This work has been taken up in a separate repository that is being actively maintained.
It is located at [fossasia/pslab-android](https://github.com/fossasia/pslab-android) , and this issue can now be closed.

closed 2017-06-08 by jithinbp