fossasia/pslab-python #43: Add relevant documentation into /docs and copy anything relevant from blog

mariobehling, 2017-04-07:
The blog is a good point to attract interest of developers. Documentation about installation etc. that can be updated should be kept in a folder /docs. 

Please copy the relevant documentation from the blog to /docs, add it as md files and update it where needed.

Keep images in /docs/img/

souravsingh, 2017-04-17:
Should we go with Sphinx docs for the documentation?

jithinbp, 2017-04-17:
I believe Python-PSLab is already set up to use Sphinx

mariobehling, 2017-04-17:
It would be good to generate the site using travis and sphinx at pslab.fossasia.org/docs/pslab-python and the other repositories in the same structure.

BassCoder2808, 2020-06-30:
@mariobehling  Do you want to create a blog site and then add the .html file into the docs or simply create a .md file and add some images into the docs folder.
It would be really helpful if you can guide me through this because I would like to contribute into this project.

closed 2020-10-19 by mariobehling