fossasia/pslab-python #45: Add Calibration Utility

jithinbp, 2017-04-14:
The PSLab unit's analog and digital features must be calibrated in order to make it accurate.
The [ADS1115 16-bit ADC](http://www.ti.com/lit/ds/symlink/ads1113.pdf) is a 4-channel, 0-3.3V ADC that can be interfaced via I2C . Typical initial accuracy of the internal voltage reference 0.01% and  data rates higher than 500SPS are possible. It is also cost effective.
Therefore, it appears to be most suited to calibrate individual PSLab devices.

Tasks
- Add I2C communication library for ADS1115
  - Include methods for reading single ended , and differential channels
  - Include selection of voltage range and voltage reference
- Add Qt based UI . Features :
  - Calibrate CH1, CH2, CH3 & PV1, PV2, PV3 using ADS1115. Generate Cubic interpolation functions for each
  - Calibrate PCS using a measured Load resistor, and calibrated analog channel. Generate interpolation function
  - Calibrate CAP using precision capacitors
  - Write all calibration constants into flash memory after assigning a timestamp
  - Store raw calibration data in a client-side folder

souravsingh, 2017-04-28:
@jithinbp Is the issue fixed? I would like to help fix the issue.

sabirma, 2017-04-28:
I'm working on this one. Got half of it covered already. 
We'll find something else for you

jithinbp, 2017-04-28:
I'm working on this one. Got half of it covered already.
We'll find something else for you

PR #46 (2017-04-14): Add support for ADS1115 16-bit ADC
 Partly Fixes #45 .
Functions included :
- Read single ended channels (0-4)
- Read differential Channels
- Select gain/voltage range
- Select Data rate
commit cecbc874 (2017-04-14): Add support for ADS1115 16-bit ADC (#46)

Partly Fixes #45 .
commit fcf6a0dd (2017-05-08): Improved auto-generated menus in graphical apps. Modified MPU6050.py. Added MPU925x.py . Fixed Sensorpendulum App & I2C oscilloscope (#57)

* Add support for ADS1115 16-bit ADC

 Partly Fixes #45 .

* modify Params dict in MPU6050.py as an example for improving sensor Menus

 refer to #55 , and https://github.com/fossasia/pslab-desktop-apps/issues/89

* MPU6050.py : remove powerUp function's argument

 refer to #55 , and https://github.com/fossasia/pslab-desktop-apps/issues/89

* MPU6050.py : Kalman Filtering's options defined in a dictionary element

 refer to #55 , and https://github.com/fos [... cut]
commit 0a017ca0 (2017-06-08): Add support for ADS1115 16-bit ADC

 Partly Fixes #45 .
closed 2017-05-09 by jithinbp