fossasia/pslab-python #49: Issues related to TSL2561.py

akarshan96, 2017-04-17:
GAIN_0X is used [here](https://github.com/fossasia/pslab-python/blob/development/PSL/SENSORS/TSL2561.py#L85) but it's value is never declared.

Also, refer  [this](https://github.com/fossasia/pslab-python/blob/development/PSL/SENSORS/TSL2561.py#L102). I think the sleeping time should be 14ms but mentioned 140ms.
It should be `time.sleep(0.014)`

jithinbp, 2017-04-17:
I think we should raise issues for most of the SENSOR classes . They were designed to do the bare minimum of acquiring data , but most sensors have a lot of additional functionality, and we need to pick and choose and implement necessary methods for configuring the sensors.

In addition , these configuration methods must be mentioned in a dictionary element called 'Params' along with a list of allowed values. The sensor data logger GUI automatically generates menu items based this dictionary.

|[image]|
|:---:|
|Figure : Auto-generated Menu for MPU6050 6-DOF IMU sensor.|

e.g. , for the HMC5883L magnetometer , the following are declared
```
    name = 'Magnetometer'
    NUMPLOTS = 3
    PLOTNAMES = ['Bx', 'By', 'Bz']
    params = {'init': ['Now'],
              'setSamplesToAverage': [1, 2, 4, 8],
              'setDataOutputRate': [0.75, 1.5, 3, 7.5, 15, 30, 75],
              'setGain': [8, 7, 6, 5, 4, 3, 2, 1],
              }
```
When the user is logging data from the magnetometer, he is also shown a menu with submenus titled 'init', setSamplesToAverage','setDataOutputRate', 'setGain' each containing their respective options.

If the user clicks on `7` in the setGain menu, the function `setGain` from the HMC5883L class is called with 7 as the argument.

This practice saves on GUI development, and adding new sensors won't require changes to the GUI. 

The getRaw function returns a list of values obtained from the sensor, and must be declared for the sensors to work with the data logger.
the NUMPLOTS variable stores the expected length of this list so that the GUI can create an appropriate number of curves
Finally, the PLOTNAMES list stores the name of each parameter in the list returned by getRaw . This allows the GUI to create a descriptive legend for the plots.

I think the above rant can be organised and added into a README file inside SENSORS submodule so that new I2C sensors can be contributed later.

akarshan96, 2017-04-18:
@jithinbp Thanks, this gave a lot of clarity.

closed 2017-05-09 by jithinbp