fossasia/pslab-python #53: "PSLab" comparison is wrong in runInitSequence function

viveksb007, 2017-04-26:
5 bytes after reading from Flash are required for comparison.
https://github.com/fossasia/pslab-python/blob/development/PSL/sciencelab.py#L183

viveksb007, 2017-04-26:
@wavicles @jithinbp would there by any other change?
As first entry is skipped in adc_slopes_offsets, so i don't think there would be any other change required.

jithinbp, 2017-04-26:
Oh, those are also leftover relics.   @akarshan96 @viveksb007 
The devices I gave you guys have the initial calibration string starting with 'SEELablet' , so I suppose you can compare against that, and it will read the calibration string properly.
Reading the calibration data is important for it to deliver accurate results . The offsets and slopes of the Op-amp stages can cause errors up to 40-50mV . This might not seem like a lot, but it might translate to significant errors when you're trying to extract physical parameters from experiments.

But I think the way the calibration constants are currently being stored can be simplified. No firmware changes will be required to do this.

>>>>   Some useful things to know regarding the current procedure :

- 20 'pages' of storage are available in flash memory.

- Each 'page' can store 2048 bytes

- The write_bulk_flash(page_number, byte_array) , read_bulk_flash(page_number, number_of_bytes_to_read ) functions can be used to access this storage space

>>>>   In the present scenario , around 9 pages are used to store rather elaborate calibration constants.

The pages , in no particular order , are :

- ADC_SLOPES_AND_OFFSETS : 1 Page . A string containing the slopes and offsets for all analog channels including the initial descriptor string .
Each analog channel's calibration is described by a 2nd degree polynomial, so that's three floats. Each float = 4bytes , so total 12 bytes per channel.
If an analog channel has an input amplifier(PGA), it will be described by 8 separate polynomials.

- ADC_INL : 2 Pages . 4KB . the integral non-linearity of the 12-bit ADC (total 4096 codes) is stored in 2 pages.
- 2 Pages for Each DAC channel. 
DACs have irregular non-linearity that cannot be corrected to a high degree of precision with a simple polynomial, so the error for each code is stored in an array. 

I've attached a picture of how a DAC's value deviates from the theoretical expectation. 

I have also attached a picture for the INL error of PIC's ADC ( This error is very small (<1mV) ), and is simply a consequence of how they're designed. For our purposes, we can simply ignore the sawtooth error, and only rectify the overall slope & offset with a 2 degree polynomial.

[image]

[image]

Cheers,
Jithin

closed 2017-05-09 by jithinbp