fossasia/pslab-python #54: Role of RPi.GPIO library in MF522.py

akarshan96, 2017-04-26:
[Here](https://github.com/fossasia/pslab-python/blob/development/PSL/SENSORS/MF522.py#L166), 22nd pin of Raspberry pi is set high. I can't get the use of the RPi.GPIO library in MF522.py.

viveksb007, 2017-04-26:
@akarshan96 PSLab device is compatible with raspberry pi too. As this is python communication library, and Pi has OS based on Linux. So this particular sensor might have been tested with Pi.

jithinbp, 2017-04-26:
Hi,
It looks like partially ported code from RPi, and the function
'MFRC522_Init()' should be deleted.
I have tested the RFID reader with this library , and the relevant
initialization registers are configured in the __init__ routine of the
Python library itself.
The RPi.GPIO import should also be removed. Please compare with the library
<https://github.com/jithinbp/SEELablet/blob/master/SEEL/SENSORS/MF522.py>
that I have verified .

The GUI also needs improvement because the RFID reader is a little slow to
respond, and the GUI should show progress indicators and such. But for now
it
can't be prioritized unless you have the reader and some smart cards.
I had a card, but I misused the rules for writing data, and wrote to
registers that apparently ruined the card.
 Hope that helps.

akarshan96, 2017-04-26:
@jithinbp Thanks, I will make required changes.

commit 760563bf (2017-06-12): Merge pull request #68 from jithinbp/development

fixed undefined references in SENSORS/MF522.py . fixes #54
commit f30396a6 (2017-06-12): fixed undefined references in SENSORS/MF522.py . fixes #54

 code ported from RPi had unused references to RPi.GPIO
closed 2017-06-12 by jithinbp