fossasia/pslab-python #61: Implementing a new feature of trigger modes

asitava1998, 2017-05-28:
##Description
We can add the various types of trigger modes found in the oscilloscopes used in Labs. They are as follows:-
A. Auto - This trigger mode allows the oscilloscope to acquire a waveform even when it does not detect a trigger condition. If no trigger condition occurs while the oscilloscope waits for a specific period (as determined by the time-base setting), it will force itself to trigger.
B. Normal - The Normal mode allows the oscilloscope to acquire a waveform only when it is triggered. If no trigger occurs, the oscilloscope will not acquire a new waveform, and the previous waveform, if any, will remain on the display.
C. Single - The Single mode allows the oscilloscope to acquire one waveform each time you press the RUN button, and the trigger condition is detected.
D. Scan – The Scan mode continuously sweeps waveform from left to right.
I can implement these in the oscilloscope UI and these would help to make it more elaborate.

jithinbp, 2017-06-06:
Currently, the indefinite wait-for-trigger option is not available. Which means only the No-trigger, Auto-trigger and scan modes are available.
I will raise this as a firmware issue.

asitava1998, 2017-06-06:
@jithinbp Yes sure. Even I was facing a problem when I was testing my code with a function generator and the PSLab as an oscilloscope.

bessman, 2021-01-18:
Moved to https://github.com/fossasia/pslab-firmware/issues/63.

closed 2021-01-18 by bessman