fossasia/pslab-python #73: x.setOffset() is missing

gabru-md, 2018-03-01:
In the file : `achan.py` on `line 128` it says
```python
x.setOffset(1.65)
```
but in the `class analogInputSource` there is no such function. Is it missing by mistake or by purpose?
Can someone clarify? If it is needed to set values of `self.defaultOffsetcode` then I can do the same.
ping @jithinbp @CloudyPadmal

bhadreshpsavani, 2019-11-05:
i think this will not create any problem because in [achan.py](https://github.com/fossasia/pslab-python/blob/development/PSL/achan.py), `line 128`  in `x.setOffset(1.65)`  is commented

@CloudyPadmal shall we close this issue or make some changes?

bessman, 2020-08-18:
Closing this since the recent achan refactorization removed this line.

closed 2020-08-18 by bessman