fossasia/pslab-python #78: Document the Python Library

CloudyPadmal, 2018-10-14:
The python library containing all the back-end functions used in PSLab has no documentation. [sciencelab.py](https://github.com/fossasia/pslab-python/blob/development/PSL/sciencelab.py) script has most of the functions and these functions need to be documented so that users can refer to them easily.

Expectation:
Prepare documentation for each function in a markdown file inside PSL directory

IgorWilbert, 2018-10-16:
I'm interested on this issue. Could you give me some model or more information?

CloudyPadmal, 2018-10-16:
Hello @IgorWilbert ! Thank you for your interest. I would recommend a model as follows;

<details>
<summary><code>Python method name with parameters</code></summary><br />

- Description of the method
- Parameters and the values they can take
  - One list item per each parameter
  - ...
- Return values and their meaning
</details>
<details>
<summary><code>Another method</code></summary><br />

- Description of the method
- Parameters and the values they can take
  - One list item per each parameter
  - ...
- Return values and their meaning
</details>
<details>
<summary><code>Another method</code></summary><br />

- Description of the method
- Parameters and the values they can take
  - One list item per each parameter
  - ...
- Return values and their meaning
</details>

Please send a pull request creating a README.md file in the PSL directory with the additions you can make. Thanks!

IgorWilbert, 2018-10-17:
@CloudyPadmal I'm trying to expand my network. Could you please follow me? I follow you back.

prateekj117, 2019-04-05:
@CloudyPadmal I would like to work on this issue.

fragm3, 2019-04-20:
@CloudyPadmal I would like to take this up.

PR #90 (2019-05-11): Document the PSL Python Library #78
Fixes: #78 
Added documentation for [sciencelab.py](https://github.com/fossasia/pslab-python/blob/development/PSL/sciencelab.py) in markdown format
closed 2019-05-11 by CloudyPadmal