fossasia/pslab-python #83: Undefined names:

cclauss, 2018-10-25:
[flake8](http://flake8.pycqa.org) testing of https://github.com/fossasia/pslab-python on Python 3.7.1

$ __flake8 . --count --select=E901,E999,F821,F822,F823 --show-source --statistics__
```
./PSL/analyticsClass.py:170:28: F821 undefined name 't'
        vf = self.sineFunc(t, par[0], par[1], par[2], par[3])
                           ^
./PSL/analyticsClass.py:171:21: F821 undefined name 'v'
        diff = sum((v - vf) ** 2) / max(v)
                    ^
./PSL/analyticsClass.py:171:41: F821 undefined name 'v'
        diff = sum((v - vf) ** 2) / max(v)
                                        ^
./PSL/analyticsClass.py:179:32: F821 undefined name 't'
            vf = self.sineFunc(t, par[0], par[1], par[2], par[3])
                               ^
./PSL/analyticsClass.py:180:25: F821 undefined name 'v'
            diff = sum((v - vf) ** 2) / max(v)
                        ^
./PSL/analyticsClass.py:180:45: F821 undefined name 'v'
            diff = sum((v - vf) ** 2) / max(v)
                                            ^
6     F821 undefined name 't'
1
```

somiljain7, 2018-10-26:
[image]

somiljain7, 2018-10-26:
can i try to work upon this issue

bhadreshpsavani, 2019-11-05:
This issue can be resolved if we pass 'v' and 't' as a parameter to the sineFit2 function in the [file](https://github.com/fossasia/pslab-python/blob/development/PSL/analyticsClass.py).

Let me work on this.

cclauss, 2019-11-05:
I added these tests to Travis CI in #112

PR #113 (2019-11-05): Undefined names issue 83
I have added t and v as a parameter of sineFit2 function in [file](https://github.com/fossasia/pslab-python/blob/development/PSL/analyticsClass.py)

This change fixes #83 issue
closed 2019-11-05 by CloudyPadmal