fossasia/pslab-python #100: Update Sphinx docs for the latest version of PSLab PCB pin layout

AakashMallik, 2019-05-18:
The docs are outdated with the names having a mismatch with the current PCB layout of PSLab.
A lack of examples in the doc also makes the use of this library hard.

bhadreshpsavani, 2019-11-06:
Hey @AakashMallik Where can we find this doc having PCB layout of PSLab

Dhriti-Roy, 2020-12-05:
@AakashMallik  May I work on this

MohitPatni0731, 2020-12-14:
Can I work on this

Radhika14soni, 2020-12-16:
Hey, I'm a first time contributor and I would like to work on this. Any guidance would be appreciated. Thanks!

Atri22, 2020-12-26:
I am a first time contributor may I work on this. Any guidance will be appreciated

prajwalbagewadi, 2021-01-23:
I to want to be a contributor any suggestions

prajwalbagewadi, 2021-01-23:
Sir can send screen shots of sphinx doc

prajwalbagewadi, 2021-01-23:
Sir is it this type of a diagram
[image]

prajwalbagewadi, 2021-01-23:
I like to work on updating docs count me in

bessman, 2021-01-23:
The docs in question can be found in https://github.com/fossasia/pslab-python/tree/development/docs, and in docstrings of individual classes and methods. The specific issue mentioned in the title was solved by https://github.com/fossasia/pslab-python/pull/141.

The documentation in https://github.com/fossasia/pslab-python/tree/development/docs is largely out of date. For now, we can remove it in favor of building the documentation entirely from docstrings. Additional documentation in the form of extended examples, working principles, and background theory could be added to https://github.com/fossasia/pslab-documentation.

closed 2021-01-23 by bessman

prajwalbagewadi, 2021-01-23:
Ok sir thanks you