fossasia/pslab-python #101: Problem while using fft funtion in the analyticClass

AakashMallik, 2019-05-22:
The **fft** funtion from the **analyticClass** of PSL has a fundamental issue that leads to **type error**

```
def fft(self, ya, si):
        '''
        Returns positive half of the Fourier transform of the signal ya.
        Sampling interval 'si', in milliseconds
        '''
        ns = len(ya)
        if ns % 2 == 1:  # odd values of np give exceptions
            ns -= 1  # make it even
            ya = ya[:-1]
        v = np.array(ya)
        tr = abs(np.fft.fft(v)) / ns
        frq = np.fft.fftfreq(ns, si)
        x = frq.reshape(2, ns / 2)
        y = tr.reshape(2, ns / 2)
return x[0], y[0]
```

The 3rd last and 2nd last line of the code should be
```
        x = frq.reshape(2, ns // 2)
        y = tr.reshape(2, ns // 2)
```

bhadreshpsavani, 2019-10-30:
Can I work on this issue?

bhadreshpsavani, 2019-10-31:
I have created [pull](https://github.com/fossasia/pslab-python/pull/109) request with the changes you suggested.

bhadreshpsavani, 2019-11-06:
Hey @CloudyPadmal 
Can you close this issue, this issue has been resolved

closed 2020-07-25 by bessman