fossasia/pslab-python #102: Refactor code base to release on pip

CloudyPadmal, 2019-06-27:
We need to give proper reformatting and documentation before we make a release on pip to make it super easy to install pslab-python

19-pragya, 2020-08-16:
Can i work on this??

bessman, 2020-08-16:
Go ahead. I'm currently refactoring the logic analyzer functionality, so if you want you can take a look at the waveform generator, the power supply, the multimeter, the motor controller, or the I2C or SPI peripherals.

19-pragya, 2020-08-16:
Ok thanks

bessman, 2020-08-16:
Some pointers:
- A lot of functionality is currently concentrated in the ScienceLab class. It's too big, so a good start is to move functionality out of there to its own module.
- We currently do not have a lot of unit tests. However, any new or refactored functionality should have unit tests written with `unittest`.
- For code style, adhere to flake8 and black, with PEP8 as the overarching style guide.
- Every public interface must be documented. Use Numpy-style docstrings, and adhere to PEP257.

19-pragya, 2020-08-16:
as this is my first time can you tell me the codebase used?

hongquan, 2020-08-16:
> any new or refactored functionality should have unit tests written with unittest

Any reason that `unittest` is preferred over `pytest`?

With `pytest`, we can incorporate other checking like `flake8`, `doctest` into one run. Code for `pytest` also look nicer with pep8-compliant function names (`setup` instead of `setUp`, `teardown` instead of `teardown`), and shorter `assert` calls.

bessman, 2020-08-16:
> Any reason that `unittest` is preferred over `pytest`?
>
> With `pytest`, we can incorporate other checking like `flake8`, `doctest` into one run. Code for `pytest` also look nicer with pep8-compliant [...]

I agree, and generally prefer pytest to unittest. However, this project relies heavily on pyserial for communication, which means we need to the serial connection during testing. For mocking, specifically, I find unittest.mock superior to pytest's monketpatch.

bessman, 2020-08-16:
> as this is my first time can you tell me the codebase used?

I'm not sure I understand your question. The codebase is what is contained in this repository. Can you be more specific?

The-Compiler, 2020-08-18:
Note you can use `unittest.mock` together with pytest as well. Either directly, or via the [pytest-mock plugin](https://github.com/pytest-dev/pytest-mock). I'd recommend the latter as it nicely integrates things with a `mocker` pytest fixture, which then undoes the patching properly without having to rely on a `unittest.mock.patch` decorator.

FWIW I personally prefer `monkeypatch` or pre-made mock libraries (e.g. [responses](https://github.com/getsentry/responses) for the requests HTTP library) because `unittest.mock` objects can be hard to configure to act like the real thing, and without proper care can allow things which wouldn't actually work against the real object.

For PySerial there doesn't seem to be such a library available in a ready-to-use state, but there are a couple of attempts which might work out:

- [ampledata/dummyserial: Dummy Serial module for mocking / testing / simulating Python Serial Interfaces.](https://github.com/ampledata/dummyserial)
- [PySerial Simulator - dftwiki](http://www.science.smith.edu/dftwiki/index.php/PySerial_Simulator)

Alternatively, socat (Linux) or [com0com](http://com0com.sourceforge.net/) can be used for end2end tests against a real (virtual) serial port, but that requires some more setup before being able to run the testsuite, so often isn't ideal either.

tl;dr: Use [pytest-mock](https://github.com/pytest-dev/pytest-mock) :slightly_smiling_face:

hongquan, 2020-08-18:
Yes, if the only reason you are tied with `unittest` is `unittest.mock`, then just use `pytest` + `pytest-mock` combo.

bessman, 2020-08-18:
Interesting, I was not aware of `pytest-mock`. Thanks for the suggestion, I'll look into it!

chrvoigt, 2020-09-25:
hi all,  using pslab-python on a  Macbook (Mojave) should be OK, right? I've installed the library as per readme.md and connected the device (through USB-C). And get an Error message (see below) ... so  it's not connecting.   Would you say it's my local setup? Or anything else I could try to get it connected? btw the same error happens if I 'npm start' the desktop app (saying 'no device connected' in the UI) 
Thanks for any pointers what to try next, 
Christian 

me@laptop ~/code/python/pslab/pslab-python (development) $ ls /dev/tty.usb*
/dev/tty.usbmodem00030777621
me@ laptop ~/code/python/pslab/pslab-python (development) $ python
Python 3.8.3 (v3.8.3:6f8c8320e9, May 13 2020, 16:29:34) 
[Clang 6.0 (clang-600.0.57)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from PSL import sciencelab
>>> I = sciencelab.connect()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/me/code/python/pslab/pslab-python/PSL/sciencelab.py", line 22, in connect
    obj = ScienceLab(**kwargs)
  File "/Users/me/code/python/pslab/pslab-python/PSL/sciencelab.py", line 86, in __init__
    self.H = packet_handler.Handler(**kwargs)
  File "/Users/me/code/python/pslab/pslab-python/PSL/packet_handler.py", line 51, in __init__
    self.interface = serial.Serial()
AttributeError: module 'serial' has no attribute 'Serial'
>>>

AbhinavMathur29pers, 2020-10-02:
can I work on the documentation and reformatting?

CloudyPadmal, 2020-10-02:
Hello @AbhinavMathur29pers , sure you can. Please go ahead :+1:

bessman, 2020-10-14:
Closing this issue since pslab-python is now on [pypi](https://pypi.org/project/pslab/). Code refactorization is still in progress and is tracked in #127.

closed 2020-10-14 by bessman