fossasia/pslab-python #116: Add unit tests

bessman, 2020-06-03:
PSL lacks unit tests. I'm interested in writing them, if you think they would be useful to the project.

mariobehling, 2020-06-03:
This would be great. We would also like to invite you to our weekly meeting at 5 pm CET on Saturdays to discuss details and find out how we could involve you more. Please join the call here https://meet.jit.si/fossasia-hw

bessman, 2020-06-05:
Thanks for the invite! I might not be able to join tomorrow though. If not, I'll join next week.

BHAKTA-VARUN, 2020-10-31:
hi everyone,
can i work with you guys
I am a beginner here.
you can assign me any task

bessman, 2020-10-31:
Hi @BHAKTA-VARUN, tests are being added as part of the ongoing refactorization effort: #127. Unfortunately it is not really beginner friendly.

Do you have a PSLab device?

BHAKTA-VARUN, 2020-11-02:
no sir i don't have  PSLab device

but thank you very much for showing concern. If in any way i can help you
please let me know.
I learnt python and mysql. If you require those skills please let me know.

Finally, please suggest any thing which I can learn to be able to
contribute.

thank you

bessman, 2021-01-18:
Closing this since most built-in instruments now have tests.

closed 2021-01-18 by bessman

viric, 2021-07-25:
Is it possible that tests for 'external' fail today? It seems there is outdated code. How do you run the tests currently?

bessman, 2021-07-26:
Tests are run with `tox -e playback` (using recorded serial traffic, requires no hardware) or `tox -e integration` (requires hardware). There are only a few tests for external right now, because as you noticed most of that code is outdated. The tests that do exist are passing.