fossasia/pslab-python #127: Split up sciencelab.py into smaller modules

bessman, 2020-07-26:
The Sciencelab class is currently almost 4000 lines long. It is actually so long that Codacy [ignores it](https://support.codacy.com/hc/en-us/articles/360025964333-File-size-limitation) when checking code quality.

As a rule of thumb, I like to keep classes under 500 lines or so. I intend to split the Sciencelab class into smaller classes based on individual instruments, like Multimeter, Oscilloscope, Waveform, LogicAnalyzer, etc. Possibly with some parent base class to avoid duplication of shared functionality.

19-pragya, 2020-08-16:
can i work on this?

bessman, 2020-08-16:
Sure. Like I said on #102, any module except for the logic analyzer is up for grabs.

vaibhavarora102, 2020-10-07:
@bessman  
i am participant for codeheat can do work on this ?

just to clarify:
we had to divide class in smaller classes and then again access them by some parent class?
am i right

vaibhavarora102, 2020-10-08:
It's having 1732 lines

bessman, 2020-10-08:
Hi @vaibhavarora102! Of course, you are welcome to work on this 😃.

I've already refactored out the oscilloscope and logic analyzer, which were about a thousand lines each. Remaining modules which should be refactored out are:

- Waveform generator (which I'm currently working on)
- Power supply
- Multimeter
- Motor controller (which depends on the waveform generator)

Refactoring a module includes the following tasks:

- Moving relevant code to a separate file, e.g. waveform_generator.py, power_supply.py, etc.
- Rewriting all code in the new file to be compliant with PEP8.
- (Re)writing documentation for all public interfaces in Numpydoc style.
- Writing integration tests with pytest.

Take a look at #129 and #132 for examples.

vaibhavarora102, 2020-10-08:
@bessman  should I continue with wave form that would be done by you
Thanks

bessman, 2020-10-08:
I'll keep working on the waveform generator. You can go ahead with either the multimeter or the power supply. Of the two, the multimeter is significantly more complicated, so it is probably a good idea for you to start with the power supply.

vaibhavarora102, 2020-10-20:
hi @bessman  
can you please clarify that in power supply their comes:
1) Motor signalling
2)READ PROGRAM AND DATA ADDRESSES
3)ANALOG OUTPUTS

or any thing else also?
or i any thing less?

thanks

bessman, 2020-10-20:
The power supply functionality consists of the methods `set_pv1`, `set_pv2`, `set_pv3`, `set_pcs`, and the associated `get_*` methods. L1292 until L1372 in the current version of sciencelab.py. These methods depend on the `DACCHAN` and `MCP4728` classes from the `Peripherals` module.

These methods and classes should be moved to a new module, `power_supply`. When moving the code, refactor it to comply with the [style guide](https://github.com/fossasia/pslab-python/blob/development/docs/development.rst#code-style). Enable linting of the new file by adding it to the `INCLUDE_PSL_FILES` environment variable of tox.ini. Finally, write integration tests and place recordings of passing tests in tests/recordings/power_supply.

vaibhavarora102, 2020-10-30:
@bessman  sir kindly unassigne the issue 
As I had already wasted much time, and still is quiet confused over this issue
Thanks

bessman, 2020-10-30:
Hi @vaibhavarora102, do you mean that you no longer wish to work on this issue, or that you would like to be assigned to it? I forgot to assign you earlier, sorry about that.

If you still want to work on this, I can try to answer any question you might have. Otherwise, no worries, thanks for giving it a try!

vaibhavarora102, 2020-10-30:
Sure if it means that I could ask as many queries as I have, then kindly assign and will work progressively

bessman, 2020-10-30:
Of course, ask away and I will do my best to answer.

bessman, 2020-11-11:
Hi @vaibhavarora102, are you still working on this?

vaibhavarora102, 2020-11-11:
Yea am working will create pull request for initial code
Within 2-3 days

vaibhavarora102, 2020-11-17:
@bessman thanks for such patience actually there was some health issue but am fine now and will now proceed with full motivation, some speed

had made initial commits and pull request 
kindly check if they are fine 
and guide for further proceeding 
I assure you to be quick this time
thanks

Skynet843, 2020-11-22:
can i do this work?

MohitPatni0731, 2020-12-14:
Can I work on this issue

mansijain980, 2021-01-10:
@bessman sir, may i start to refactor "Power supply " module ? I am a participant in Codeheat . I can make initial split for power supply. I got how to divide sciencelab.py module into smaller classes.
 
Please assign this issue to me.
It will be a nice start for me.
Thanks

bessman, 2021-01-10:
@mansijain980 Thanks for the offer, but I am already working on it.

mansijain980, 2021-01-10:
I already started working on that but it's fine. I would like to work on "Multimeter" module then.

Can you assign this to me ?

mansijain980, 2021-01-10:
@bessman sir,
I already started working on that but it's fine. I would like to work on "Multimeter" module then.

Can you assign this to me ?

bessman, 2021-01-11:
The multimeter refactorization is done. Feel free to open a pull request with the changes that you have so far, perhaps we can incorporate them somehow.

SAP-20, 2021-01-15:
A newbie in the open source contributing.Sir,Could you assign me this project to me??

bessman, 2021-01-17:
Hi @SAP-20. We are moving away from assigning issues, and instead encourage contributors to work on whatever interests them, and to open pull requests whenever they are ready to discuss their changes.

This particular issue is just about finished, as of #158. If any of the other open issues seem interesting to you, please go ahead and work on any of them you'd like. No need to wait to be assigned :smile: 

One of the most valuable ways to contribute to pslab-python right now is to simply use the library. We now have automated tests for most of it, but automated tests can't compete with actual users when it comes to finding bugs. Right now, bug reports are probably more valuable to us than code contributions.

bessman, 2021-01-18:
Most of the useful stuff in sciencelab.py has now been moved to separate modules. It is over 3000 lines of code shorter than it was when this issue was opened. Most of what remains in sciencelab.py is unused and/or broken and should be removed.

closed 2021-01-18 by bessman