fossasia/pslab-python #133: Resource manager

bessman, 2020-08-30:
Certain functions share the same hardware resource. For example, the oscilloscope and the logic analyzer both use the ADC buffer, which means they cannot be used at the same time. Another example is the pulse counter, which shares a timer with the logic analyzer.

We should create a resource manager to keep track of this, to prevent users from accidentally using multiple functions which use the same hardware simultaneously. Doing so can yield extremely strange results.

gr8geek, 2020-09-01:
@bessman I am willing to work on this issue

shrey-3108, 2020-09-24:
i would like to work on that

bessman, 2020-09-24:
Feel free. To start with, we need to compile a list of available hardware resources. Then, we will need a list of which firmware features use which hardware resource.

Be advised that this task is not beginner friendly. It requires a deep understanding of the PSLab on the software, firmware, and hardware level. Prerequisite skills are electronics and hardware development, C, and Python.

Maxz2136, 2020-10-01:
I am new to this project, but upon reading this issue, I wonder can locking mechanism like semaphore help achieve the synchronization between firmware features using the same hardware resource?

MohitPatni0731, 2020-12-14:
Can I work on this issue

pratik-99, 2021-01-22:
Hello @bessman can you suggest me where I could learn about the software, firmware and hardware of PSlab I would like to contribute here if I have a general Idea about it

bessman, 2021-01-23:
Best way to learn about the hardware is to read the schematics at https://github.com/fossasia/pslab-hardware, as well as datasheets for the individual components.

Firmware, read the source code on https://github.com/fossasia/pslab-firmware.

Software is the current repo.

However, we are probably going to upgrade the PSLab's MCU in the near future. A resource manager written for the current hardware would very likely not carry over to the new one, so working on this issue right now would be wasted effort. Therefore, I will temporarily close this issue. Once we have decided on a new MCU and released a new revision of the PSLab, we can consider reopening it.

closed 2021-01-23 by bessman