fossasia/pslab-python #135: Connecting PSLab (Macbook)

chrvoigt, 2020-09-25:
hi all, using pslab-python on a Macbook (Mojave) should be OK, right? I've installed the library as per readme.md and connected the device (through USB-C). And get an Error message (see below) ... so it's not connecting. Would you say it's my local setup? btw the same error happens if I 'npm start' the desktop app (saying 'no device connected' in the UI)
Thanks for any pointers as to what else I should try,
Christian

me@laptop ~/code/python/pslab/pslab-python (development) $ ls /dev/tty.usb*
/dev/tty.usbmodem00030777621
me@ laptop ~/code/python/pslab/pslab-python (development) $ python
Python 3.8.3 (v3.8.3:6f8c8320e9, May 13 2020, 16:29:34)
[Clang 6.0 (clang-600.0.57)] on darwin
Type "help", "copyright", "credits" or "license" for more information.

from PSL import sciencelab
I = sciencelab.connect()
Traceback (most recent call last):
File "", line 1, in
File "/Users/me/code/python/pslab/pslab-python/PSL/sciencelab.py", line 22, in connect
obj = ScienceLab(**kwargs)
File "/Users/me/code/python/pslab/pslab-python/PSL/sciencelab.py", line 86, in init
self.H = packet_handler.Handler(**kwargs)
File "/Users/me/code/python/pslab/pslab-python/PSL/packet_handler.py", line 51, in init
self.interface = serial.Serial()
AttributeError: module 'serial' has no attribute 'Serial'

bessman, 2020-09-25:
You may have installed [serial](https://pypi.org/project/serial/) instead of [pyserial](https://pypi.org/project/pyserial/).

chrvoigt, 2020-09-25:
yes indeed - thanks for the hint !

I had both installed and there was a namespace conflict .. 
So a dedicated virtualenv did the trick and then 'pip freeze' confirmed that PSL and pyserial are included. 

... and now the electron app is running too 👍

I changed the title (since the issues was about python modules),  in case someone experiences this as a first hurdle when trying to connect.

closed 2020-09-25 by chrvoigt