fossasia/pslab-python #140: Update pin names

bessman, 2020-10-05:
Change all instances of

- ID1/2/3/4 -> LA1/2/3/4
- SEN -> RES
- AN8 -> VOL
- CNTR -> FRQ
- W1/2 -> SI1/2

MLJBrackett, 2020-10-05:
Hey @bessman I would like to do this, could you assign it to me?

Nanak360, 2020-10-07:
I would like to work on it. It will be very helpful if I could get a little more information. 
[image]
[image]

I see many instances of ID1, ID2, SEN
Should I change them all in each and every file?

bessman, 2020-10-07:
Thanks! Yes, that's right. Every instance in every file.

Nanak360, 2020-10-07:
All right, I'll work on it. Would you please assign me?
I would also like to know if this contribution will be a part of Hacktoberfest.

bessman, 2020-10-07:
Sure, we can opt in and tag your pull request as hacktoberfest-approved.

Nanak360, 2020-10-07:
@bessman I have changed each and every instance from all the files and created a pull request. Please guide me through, what do I do now for the Hacktoberfest part.

bessman, 2020-10-07:
Fixed by #141.

closed 2020-10-07 by bessman