fossasia/pslab-python #142: explain installing pslab-python to the system and setting PYTHONPATH

orangecms, 2020-10-07:
see also https://github.com/fossasia/pslab-documentation/pull/119#issuecomment-705048446

For PSLab desktop app users, especially on Windows, connecting pslab-desktop to pslab-python needs some more information.

bessman, 2020-10-07:
I wonder if we could simplify the windows instructions. Theoretically, the Linux instructions should actually be platform independent.

We should test if
```pip install git+https://github.com/fossasia/pslab-python@master
```
works on Windows and Mac. If it does, we could cut most of the current installation instructions entirely.

bessman, 2020-10-14:
As pslab-python is now on [pypi](https://pypi.org/project/pslab/), installing should now be as simple as `pip install pslab`, regardless of operating system.

closed 2020-10-14 by bessman