fossasia/pslab-python #145: Handling scenarios where no devices are connected

CloudyPadmal, 2020-10-14:
When a user tries the PSL library without any device connected, the following error message is thrown.

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/padmal/.local/lib/python3.6/site-packages/PSL/sciencelab.py", line 22, in connect
    obj = ScienceLab(**kwargs)
  File "/home/padmal/.local/lib/python3.6/site-packages/PSL/sciencelab.py", line 86, in __init__
    self.H = packet_handler.Handler(**kwargs)
  File "/home/padmal/.local/lib/python3.6/site-packages/PSL/packet_handler.py", line 65, in __init__
    self.connect(port=port, baudrate=baudrate, timeout=timeout)
  File "/home/padmal/.local/lib/python3.6/site-packages/PSL/packet_handler.py", line 161, in connect
    raise serial.SerialException("Device not found.")
serial.serialutil.SerialException: Device not found.
```

@bessman May be we better handle it gracefully?

bessman, 2020-10-14:
What behavior would you prefer? pslab-python currently offers no functionality without a connected device, so a "Device not found" exception seems fine to me.

CloudyPadmal, 2020-10-14:
I was thinking something like just the error message without the whole error log :thinking:

bessman, 2020-10-14:
I can see two ways to accomplish that, neither of which I like:

1. Catch the exception and print the error message: `except SerialException as e: print(e)`
Exceptions should not be caught without good reason, and wanting to show the user a nicer looking error message is not a good enough reason.
2. Reduce traceback depth: `import sys; sys.tracebacklimit = 0`
Less bad, but why only here? Should we suppress the traceback throughout the whole library? That would be a terrible idea, and I don't see why this exception deserves special treatment.

If we want to enhance the user experience, we should create a simple CLI application. Adding this kind of hack to the base library only complicates troubleshooting.

bessman, 2020-10-14:
I opened an issue to create a CLI: #146 

Do you think that is an OK solution?

CloudyPadmal, 2020-10-14:
Yes, I think that would be a good approach :+1: I'll close this issue to continue.

closed 2020-10-14 by CloudyPadmal