fossasia/pslab-python #146: Create a CLI application

bessman, 2020-10-14:
Some users might now need the full GUI suite offered by pslab-desktop, but might still want a nicer UX than just typing into the Python REPL. We should create a CLI application that can handle certain common tasks, like collecting samples with the oscilloscope, outputting a waveform, etc.

nkpro2000sr, 2020-10-14:
can i work on this?

bessman, 2020-10-14:
Please do! 😃

bessman, 2020-10-14:
@nkpro2000sr Please also take a look at our new [contributor guidelines](https://github.com/fossasia/pslab-python/blob/development/docs/development.rst).

nikhiltidke101, 2020-10-15:
I would like to work on this. Can I?

bessman, 2020-10-15:
Hi @nikhiltidke101. nkpro2000sr has already been assigned this task, and it is not really suitable for multiple people to work on it simultaneously.

We are currently a little short on beginner-friendly tasks, but if you would like to help out I would recommend just starting to use the PSLab and help fix any bugs you encounter. Something is bound to crop up 😃

orangecms, 2020-10-15:
I have some examples here for usage, might help a bit: https://github.com/orangecms/pslab-python/tree/examples/examples

nkpro2000sr, 2020-10-16:
i have a doubt that, is collecting data from the oscilloscope and logic analyzer has to be collected **until timeout** or **certain interrupt like pressing <kbd>q</kbd>**?

orangecms, 2020-10-16:
There are many possibilities like offering an argument to specify a time period. I could imagine subcommands like many modern CLIs offer them, e.g., `pslab collect --instrument oscilloscope --duration 3s --output my-data.json`. What do you think? :)

nkpro2000sr, 2020-10-16:
nice :+1:.
while plotting which will be nicer?
while collecting (in live) OR after collected

orangecms, 2020-10-16:
You can always plot in various ways afterwards. Keep data separate and reusable. :) The CLI also allows for scripting on a higher level, in a Unix way, so you can combine with other tools through pipes, for example.

We have the desktop app for live plotting already, so there is no need to duplicate that.

nkpro2000sr, 2020-10-16:
suggestions please?
```
$ python PSL --help
usage: PSL [-h] [-p PORT] {collect} ...

optional arguments:
  -h, --help            show this help message and exit
  -p PORT, --port PORT  The name of the port to which the PSLab is connected

Functions:
  Functions to perform on PSLab

  {collect}

$ python PSL collect --help
usage: PSL collect [-h] -i INSTRUMENT [-c CHANNELS] [-d DURATION] [-o FILE_PATH] [-s]

optional arguments:
  -h, --help            show this help message and exit
  -i INSTRUMENT, --instrument INSTRUMENT
                        The name of the instrument to use
  -c CHANNELS, --channels CHANNELS
                        Number of channels to capture
  -d DURATION, --duration DURATION
                        Duration for capturing (in seconds)
  -o FILE_PATH, --output FILE_PATH
                        File name to store data as csv file
  -s, --stdout          Enable it to output data as json in stdout

$ python PSL -p /dev/USB0 collect --instrument oscilloscope --duration 3 --output my-data.csv 

$ python PSL -p /dev/USB0 collect --instrument logic_analyzer --channel 2 --stdout > my-data.json
```

bessman, 2020-10-16:
Very nice! The only thing I would add to the help text is a list of available instruments.

orangecms, 2020-10-16:
Really cool!
For intuitive use, I would suggest to always print to stdout unless a `-o` option for an output file is given, and default to linewise CSV unless a `-j` / `--json` switch is provided. The allows for any combination thereof then.

orangecms, 2020-10-16:
Another little thing: Commands are usually all lowercase. There is already a `psl` command https://docs.bmc.com/docs/PATROLAgent/110/psl-psl-compiler-command-723218465.html#:~:text=The%20psl%20command%20creates%20an,can%20be%20used%20in%20the%20. so I would suggest to go with the full name `pslab`. I will make sure that the name for the desktop app will be `pslab-desktop`, which makes it similar to `signal-desktop` and other similar apps.

nkpro2000sr, 2020-10-16:
But, I have planned to create PSL/__main__.py. so, it will be like `python -m PSL ...`.
Also it didn't required any extra setup, just `pip install pslab` would be enough.

bessman, 2020-10-16:
You can add the CLI application as either a `scripts` or `console_scripts` entry in setup.py. That way, it can be run directly from the command line as `pslab`, without `python -m ...`.

https://python-packaging.readthedocs.io/en/latest/command-line-scripts.html

nkpro2000sr, 2020-10-16:
thanks, i didn't aware of that.

nkpro2000sr, 2020-11-21:
suggestions please?
```bash
[nkpro@h0me pslab-python]$ pslab -h
usage: check.py [-h] [-p PORT] {collect,wave} ...

optional arguments:
  -h, --help            show this help message and exit
  -p PORT, --port PORT  The name of the port to which the PSLab is connected

Functions:
  Functions to perform on PSLab.

  {collect,wave}
```
```bash
[nkpro@h0me pslab-python]$ pslab wave -h   
usage: check.py wave [-h] {gen,load} ...

optional arguments:
  -h, --help  show this help message and exit

Wave Functions:
  {gen,load}
```
```bash
[nkpro@h0me pslab-python]$ pslab wave gen -h
usage: check.py wave gen [-h] -f FREQUENCY [FREQUENCY ...] [-p PHASE] {SI1,SI2} [{SI1,SI2} ...]

positional arguments:
  {SI1,SI2}             Pin(s) on which to generate a waveform

optional arguments:
  -h, --help            show this help message and exit
  -f FREQUENCY [FREQUENCY ...], --frequency FREQUENCY [FREQUENCY ...]
                        Frequency in Hz
  -p PHASE, --phase PHASE
                        Phase between waveforms in degrees
```
```bash
[nkpro@h0me pslab-python]$ pslab wave load -h
usage: check.py wave load [-h] (--table TABLE | --table-file [TABLE_FILE]) {SI1,SI2}

positional arguments:
  {SI1,SI2}             Pin(s) on which to load a table

optional arguments:
  -h, --help            show this help message and exit
  --table TABLE         Table to load in pin SI1 as json
  --table-file [TABLE_FILE]
                        Table to load in pin SI1 as json file. Default is stdin
```
```bash
[nkpro@h0me pslab-python]$ pslab wave gen SI1 SI2 -f 50 60 -p 10
```
```bash
[nkpro@h0me pslab-python]$ pslab wave load SI1 --table [0,1.65,3.3,1.65,0,-1.65,-3.3,-1.65,0]
```
```bash
[nkpro@h0me pslab-python]$ cat table.json | pslab wave load SI1 --table-file
```
```bash
[nkpro@h0me pslab-python]$ pslab wave load SI1 --table-file table.json
```

orangecms, 2020-11-21:
Is there a branch we can check out to play around with it? So far it looks quite nice to me :)

bessman, 2020-11-21:
PRs can be checkout out directly:
```bash
git fetch upstream pull/147/head:cli
git checkout cli
```

@nkpro2000sr The syntax looks good! I'll try it out and come back with more feedback.

orangecms, 2020-11-21:
Yea sorry, I missed the draft PR, thank you!

Skynet843, 2020-11-22:
is Issue still open can i work on that?

nkpro2000sr, 2020-11-28:
> @nkpro2000sr The syntax looks good! I'll try it out and come back with more feedback.

Any feedback ... :eyes:

bessman, 2020-11-28:
Yeah, sorry, been kind of swamped this week. Feedback coming up.

bessman, 2020-11-28:
All right, first and foremost, thanks for your patience!

I'll go through the commands in order:

- collect
    - logic_analyzer
        Seems to work well! :+1: 
    - oscilloscope
        This one doesn't seem to work. I get no output, either to stdout or to a file with -o.

- wave
    - gen
        Seems to work well! :+1:
    - load
        A table format example would we useful here.

- pwm
    - gen
        Seems to work well! :+1:
    - map
        I get `struct.error: required argument is not an integer`. Is the prescaler argument interpreted as a float maybe?
    - set
        Not really necessary I think, since the same thing can be accomplished by setting the duty cycle to 0 or 1 with gen.

nkpro2000sr, 2020-11-28:
@bessman 
fixed :+1:

bessman, 2020-11-28:
Thank you. `pwm map` works well now, and the format example in `wave load` is good. `collect oscilloscope` is still problematic though; it runs for much longer than the set duration. With `--duration 1` it runs for almost a minute.

nkpro2000sr, 2020-11-28:
It collects 2e6 samples for a second (-d 1).
I doubt that delay is due to massive samples fetched.

could you please check `scope.capture(1, 2e6, 0.5)` ?

bessman, 2020-11-28:
The oscilloscope can't capture more than 1e4 samples at a time, so `scope.capture(1, 2e6, 0.5)` will just complain about that. There is a fair bit of overhead, time wise, between individual capture calls, so capturing 2e6 samples will take a lot longer than just 2e6 * 0.5 * 1e-6.

nkpro2000sr, 2020-11-28:
Would increasing time_gap will help?

nkpro2000sr, 2020-11-28:
would [this](https://github.com/fossasia/pslab-python/pull/147/commits/e077148570b26772b8d03b430fb181937d0ac286) solves?
```diff
+        st = time.time()
         xy = np.append(xy, scope.capture(channels, samples, min_timegap), axis=1)
-        duration -= max_duration
+        duration -= time.time() - st
```

But this would miss some samples.

bessman, 2020-11-28:
Yes, that should work. I'll test it tomorrow. Do you not have a device to test with?

Missing samples is unavoidable; after each capture, the 10000 samples (20 kB) must be transferred across the 1 Mbaud serial connection, which takes 160 ms. During that time, no samples can be collected.

nkpro2000sr, 2020-11-28:
> Yes, that should work. I'll test it tomorrow.  

:+1: 

> Do you not have a device to test with?  

No, i don't have :(

> Missing samples is unavoidable; after each capture, the 10000 samples (20 kB) must be transferred across the 1 Mbaud serial connection, which takes 160 ms. During that time, no samples can be collected.  

ok :+1:

bessman, 2020-11-29:
It works now, well done! I'm impressed you managed to do so much without have the hardware to test against.

Do you think you could also write some tests for the cli module?

nkpro2000sr, 2020-11-30:
@bessman 
Added tests for CLI. please check.

bessman, 2020-12-02:
Thanks!

Could you also write tests for the `logic_analyzer` and `oscilloscope` functions? The other missed lines are OK because they just use already tested interfaces, but those functions contain logic that should ideally be covered.

And one more thing: We are moving from unittest to pytest, so please use pytest-mock instead of unittest.mock. It should be a drop-in replancement.

nkpro2000sr, 2020-12-04:
@bessman 
Added testes for `logic_analyzer` and `oscilloscope`.

Could you add json traffic recordings for these. Since i don't have device i am not able to generate those json fils.

bessman, 2020-12-29:
Thank you @nkpro2000sr, your work has been merged!

closed 2020-12-29 by bessman