fossasia/pslab-python #159: Bug: CH1 and CH2 autorange set incorrectly

cynthi8, 2021-01-23:
1. Reading a measurement from CH1 or CH2
2. Removing the measurement probes
3. Re-inserting them 

yields incorrect results.

When `_voltmeter_autorange`is bypassed by calling `_measure_voltage` directly, this issue does not occur.

**Setup**
I connected CH1 measurement pins to I2C VDD (3.3 V) and GND and ran this code
```
import time
import sys
from PSL import sciencelab
I = sciencelab.connect()

channel = sys.argv[1]

while True:
    voltage = I.multimeter.measure_voltage(channel)
    print(channel, "=", voltage)
    time.sleep(1)
```
**Results**
[image]

With `_measure_voltage` instead of `measure_voltage`,
[image]

bessman, 2021-01-23:
Fixed in cc9ad37.

closed 2021-01-23 by bessman

cynthi8, 2021-01-23:
Ah ok, sorry, I didn't check. Thanks!

bessman, 2021-01-23:
No worries. Good job on finding the bug either way. I'll draft a new release and push it to pypi shortly.