fossasia/pslab-python #170: concurrent USB devicss

orangecms, 2021-02-25:
Cross-referenced from pslab-desktop:
https://github.com/fossasia/pslab-desktop/issues/673

Please have a look over there, an issue reported with multiple USB serial devices present.

bessman, 2021-02-25:
This is already possible by specifying the port name of the second device, e.g:

```python
import pslab
h1 = pslab.serial_handler.SerialHandler(port="/dev/ttyUSB0")
h2 = pslab.serial_handler.SerialHandler(port="/dev/ttyUSB1")
scope1 = pslab.Oscilloscope(h1)
scope2 = pslab.Oscilloscope(h2)
```

However, autodiscovery currently only picks up the first device. That's worth fixing.

SwapnilSoni1999, 2021-02-27:
So what is the expected behaviour?

orangecms, 2021-02-28:
Here's my suggestion:
Iterate over all USB devices, see if they respond with a PSLab name / version number, and in the desktop app, offer a list. The CLI could report that there are multiple devices available and ask to rerun the command with the device specified through an argument.
That is also the behvior of other tools, e.g., `flashrom`.

What do you think?

bessman, 2021-02-28:
That seems like a good solution.

bessman, 2021-02-28:
#173

bessman, 2021-03-04:
I'm closing this since #173 is merged. Let me know if you think the solution works for pslab-desktop.

closed 2021-03-04 by bessman

orangecms, 2021-03-04:
Sounds good, thank you so much! :)