fossasia/pslab-python #171: Bug: Autorange not working for oscilloscope

cynthi8, 2021-02-26:
**Issue:**
1. Reading from the multimeter
2. Reading from the oscilloscope
returns incorrect values.

**Test Code**
```
from pslab import ScienceLab
psl = ScienceLab()
voltage = psl.multimeter.measure_voltage("CH1")
print(voltage)
oscilloscope_capture = psl.oscilloscope.capture("CH1", 1, 1)
print(oscilloscope_capture)
```

**Result**
```
3.3352564102564104
[array([0.]), array([13.33296703])]
```

I would expect the oscilloscope to return a value around 3.3 V.

cynthi8, 2021-02-26:
Ok, go for it! Should be a quick fix. Checking out #159 might give you some context on how previous auto gain setting bugs were fixed.

bessman, 2021-02-26:
@Kartik125 Go ahead. The problem is that the state of the analog input gain is not shared between different instances that use the analog inputs. The easiest way to fix this is for every Oscilliscope instance to always set the most recently selected gain before starting capture.

A more robust solution would be to make AnalogInput into a shared multiton, but that would require significantly more work and the extra complexity may or may not be worth it.

Kartik125, 2021-02-26:
I'll keep in mind

Kartik125, 2021-02-27:
can you guide me where can I find this issue I'm new at this, I will be pleased if you tell me

bessman, 2021-02-27:
The issue is caused by the actual gain being different from what the oscilloscope instance thinks it is. Since there is no way to query the device to find out the gain, each oscilloscope instance must keep track of what the gain is. If there are multiple oscilloscope instances, one of them can change the gain without the other instances knowing about it. The next time such an instance captures samples, it might think that the gain is, for example, 1 when it is really 4. Then the returned values will be off by a factor of 4.

In order to solve that, each oscilloscope instance should first set the gain to whatever it thinks it is before capturing samples.

On [L134](https://github.com/fossasia/pslab-python/blob/development/pslab/instrument/oscilloscope.py#L134) in oscilloscope.py, before the `self._capture` call, call `self._set_gain` for each channel in `self._channels` and set the gain to the value stored in `channel.gain`.

Kartik125, 2021-02-27:
Thanks, I'll check it out.

bessman, 2021-03-06:
Any progress on this, @Kartik125?

Kartik125, 2021-03-07:
Yes, I am working on it.

commit 3f3e5191 (2021-03-15): Set gain before capturing trace (fix #171)
closed 2021-03-15 by bessman