fossasia/pslab-python #183: Migrate from Travis CI

orangecms, 2021-05-22:
We need to take of migrating, it says

> Please be aware travis-ci.org will be shutting down by end of May 2021. Please consider migrating to travis-ci.com.

That is a week left only :scream: 

see e.g. https://travis-ci.org/github/fossasia/pslab-python/builds/772004871

orangecms, 2021-06-05:
Update: They have extended the deadline to June 15.

CloudyPadmal, 2021-06-06:
Let's migrate to GitHub actions 🙌🏼

bessman, 2021-06-06:
That didn't work as well as I had hoped. It seems Github Actions are disabled for this repository. I don't have sufficient permissions to enable them.

orangecms, 2021-06-06:
Yes, configuration is necessary first. Otherwise people could do all sorts of fun stuff just by opening PRs. @mariobehling could you provide that and/or grant the access rights? :)

CloudyPadmal, 2021-06-06:
Looks like me neither.. @mariobehling can you please check?

commit 80f43064 (2021-06-06): chore: Migrate to Github Actions (closes #183)
closed 2021-06-07 by bessman