fossasia/pslab-python #197: Implement missing SQR1 and SQR2 functions

CloudyPadmal, 2021-06-24:
In the previous psl package, there were functions to control SQR1 and SQR2 pins individually. However, they are missing in the latest version and one has to use SQR4 function driving all 4 SQR pins.

With the latest firmware upgrades, it is made possible to generate sine and square waves simultaneously and we need to implement whatever the missing functions referring to the old python package.

Refer to following as a guide:
- [SQR1](https://github.com/fossasia/pslab-python/blob/420a97eaac0f54d5185ad785e09a205340f422df/PSL/sciencelab.py#L3127)
- [SQR2](https://github.com/fossasia/pslab-python/blob/420a97eaac0f54d5185ad785e09a205340f422df/PSL/sciencelab.py#L3197)

bessman, 2021-06-24:
We will need to add a way to check the firmware version as well, because I think WAVEGEN->SET_SQR2 is broken in the current version.

CloudyPadmal, 2021-06-24:
That's true.. in the old SQR2 function in firmware, the right timer is never turned on making it broken. I wonder if there's a way to trigger it indirectly. 🤔

Now that I'm thinking, it would be nice if we had a getter setter function to access any arbitrary register in firmware.

marcnause, 2025-12-29:
@bessman Is this issue still valid?