fossasia/pslab-python #200: External device meta data

bessman, 2021-07-02:
Devices in `pslab.external`, such as sensors, motors, and displays, should include some kind of meta data so that front ends like pslab-desktop and pslab-cli can use them without large amounts of custom code.

One possible solution is a JSON file for every supported device, with a structure like this:
```json
{
"name": "MLX90614",
"modes":
    {
    "thermometer":
        {
        "input": 
            {
            "name": "source",
            "type": "selection",
            "options": ["object", "ambient"],
            },
        "output": 
            {
            "name": "temperature",
            "type": "float", 
            "range": [-127, 128],
            "unit": "degC",
            },
        },
    },
}
```
This file specifies that the MLX90614 class has:
- A single mode called "thermometer".
- A `set` method which accepts "source" followed by either `"object"` or `"ambient"` as its argument. 
- A `get` method which accepts "temperature" and returns a float between -127 degC and 128 degC.
 
A more complex example:
```json
{
"name": "BMP180",
"modes":
    {
    "thermometer":
        {
        "output": 
            {
            "name": "temperature",
            "type": "float", 
            "range": [-127, 128],
            "unit": "degC",
            },
        "default": 1,
        },
    "pressure meter":
        {
        "output": 
            {
            "name": "pressure",
            "type": "float", 
            "range": [0, 1e7],
            "unit": "Pa",
            },
        "default": 0,
        },
    "altitude meter":
        {
        "input":
            {
            "name": "baseline",
            "type": "float",
            "range": [0, 1e7],
            "unit": "Pa",
            },
        "output": 
            {
            "name": "altitude",
            "type": "float", 
            "range": [-1e2, 1e4],
            "unit": "m",
            },
        "default": 0,
        },
    },
}
```
This file specifies that the BMP180 class has:
- A multiple modes, which can be selected by setting the `BMP180.mode` property, the default being "thermometer".
- No setters in the "thermometer" or "pressure meter" modes.

By reading these files, the front end does not need to know any details about the sensors and requires no custom code. @orangecms what do you think?

nkpro2000sr, 2021-07-03:
Also Adafruit's drivers which uses CircuitPython-busio can be included with external sensors, since pslab-python has busio now.

Example:
```python3
# pslab/external/busio/si7021.py
import adafruit_si7021
from pslab.bus import busio

class SI7021(adafruit_si7021.SI7021):
    _ADDRESS = 0x40

    def __init__(self, **args):
        self._ADDRESS = args.get('address', self._ADDRESS)
        self._i2c = args.get('i2c_buss', busio.I2C())
        super().__init__(self._i2c, self._ADDRESS)
```

```json
{
"name": "SI7021",
"modes":
    {
    "thermometer":
        {
        "output": 
            {
            "name": "temperature",
            "type": "float", 
            "range": [-10, 85],
            "unit": "degC",
            },
        "default": 1,
        },
    "moisture meter":
        {
        "output": 
            {
            "name": "relative_humidity",
            "type": "int", 
            "range": [0, 80],
            "unit": "% RH",
            },
        "default": 0,
        },
    },
"dependencies" :
    [
        "adafruit-circuitpython-si7021",
    ],
}
```
> We also need "dependencies" to check/install/manage dependencies for these sensors.

Shubham-Padkonde, 2026-10-27:
Is this metadata direction still wanted? I'd like to take a small first slice if it is available.

I checked the current sensor interfaces before proposing a format. `BMP180` now exposes `temperature`, `pressure` and `altitude` properties, while `MLX90614` has methods such as `getObjectTemperature()` and `getAmbientTemperature()`. The generic `mode`/`get`/`set` interface in the original examples would therefore need either an adapter or explicit mappings to the existing API.

My proposed first scope is a documented descriptor format plus one BMP180 example, covering the public member names, units, read/write access and optional dependencies mentioned above. Metadata discovery should work without constructing a sensor or connecting to a board. I would add validation tests for the descriptor and its declared API members, while leaving sensor behavior and the other drivers unchanged.

Would you prefer descriptors to map the current properties/methods, or a common adapter interface? If a descriptor-first slice is useful, could a maintainer assign that scope to me? I'll wait for that decision before implementing it. I don't have a physical PSLab board, so this would validate metadata and discovery only, not sensor measurements.