fossasia/pslab-python #206: Increase the usage of augmented assignment statements

elfring, 2021-11-02:
:eyes: Some source code analysis tools can help to find opportunities for improving software components.
:thought_balloon: I propose to [increase the usage of augmented assignment statements](https://docs.python.org/3/reference/simple_stmts.html#augmented-assignment-statements "Augmented assignment statements") accordingly.

```diff
diff --git a/pslab/external/MF522.py b/pslab/external/MF522.py
index 1b30454..8c25bc4 100644
--- a/pslab/external/MF522.py
+++ b/pslab/external/MF522.py
@@ -250,7 +250,7 @@ class MF522:
         i = 2000
         while True:
             n = self.read(self.ComIrqReg)
-            i = i - 1
+            i -= 1
             if ~((i != 0) and ~(n & 0x01) and ~(n & waitIRq)):
                 break
 
@@ -279,7 +279,7 @@ class MF522:
                     i = 0
                     while i < n:
                         returnedData.append(self.read(self.FIFODataReg))
-                        i = i + 1;
+                        i += 1;
             else:
                 status = self.MI_ERR
         return (status, returnedData, backLen)
@@ -315,8 +315,8 @@ class MF522:
             i = 0
             if len(returnedData) == 5:
                 while i < 4:
-                    serNumCheck = serNumCheck ^ returnedData[i]
-                    i = i + 1
+                    serNumCheck ^= returnedData[i]
+                    i += 1
                 if serNumCheck != returnedData[i]:
                     status = self.MI_ERR
             else:
@@ -347,7 +347,7 @@ class MF522:
         i = 0
         while i < 5:
             buf.append(serNum[i])
-            i = i + 1
+            i += 1
         pOut = self.CalulateCRC(buf)
         buf.append(pOut[0])
         buf.append(pOut[1])
@@ -368,13 +368,13 @@ class MF522:
         i = 0
         while (i < len(Sectorkey)):
             buff.append(Sectorkey[i])
-            i = i + 1
+            i += 1
         i = 0
 
         # Next we append the first 4 bytes of the UID
         while (i < 4):
             buff.append(serNum[i])
-            i = i + 1
+            i += 1
 
         # Now we start the authentication itself
         (status, returnedData, backLen) = self.MFRC522_ToCard(self.PCD_MFAuthent, buff)
@@ -422,7 +422,7 @@ class MF522:
             buf = []
             while i < 16:
                 buf.append(writeData[i])
-                i = i + 1
+                i += 1
             crc = self.CalulateCRC(buf)
             buf.append(crc[0])
             buf.append(crc[1])
@@ -441,7 +441,7 @@ class MF522:
                 self.MFRC522_Read(i)
             else:
                 print("Authentication error")
-            i = i + 1
+            i += 1
 
 
 if __name__ == "__main__":
```

bessman, 2021-11-02:
Sure, that's a good idea. However, this particular module (MF522) is most likely broken since no active developer has access to that sensor. We can only accept a pull request with this kind of improvement if it also fixed the module itself.

elfring, 2025-04-06:
>This pull request addresses …

Where was this contribution submitted? :thinking:

Nivriti-ctrl, 2025-04-07:
Sir It didn't show commit changes in this issue.

bessman, 2025-04-07:
@Nivriti-ctrl Stop adding LLM generated comments on old issues.