fossasia/pslab-python #212: Trigger setting not working in Pslab v5 board

viric, 2022-11-17:
I'm trying a very simple set up, with an electrical signal in CH1, trigger in CH1 at 1V, and the enabling the "trigger" setting makes no difference. It keeps on updating the display as if it were not checked.
I tested the Android and the pslab-desktop app, and also the pslab-python alone (with the oscilloscope capture() trigger parameters and also with configure_trigger()).
Maybe it's a bug in the PSlab v5 board or firmware? I bought the board some years ago. I don't know if I can check or update the board firmware easily.

bessman, 2022-11-17:
Yeah, the trigger can be a little finicky. Do you know roughly what the signal looks like, and what its frequency is? One possible reason for the trigger not working is that it times out waiting for the trigger condition to be met. The timeout is 6.25 ms, so if your signal has a frequency less than 200 Hz or so the trigger might be timing out.

Check if the trigger works on a signal generated by the PSLab itself. Connect pin SI1 to CH1, and run this code:

```python
import matplotlib.pyplot as plt
import pslab

psl = psl.ScienceLab()
psl.waveform_generator.generate(1, 1000)  # 1 kHz sinusoid on SI1.
for _ in range(5):
    x, y = psl.oscilloscope.capture(1, 1000, 10, trigger=1)  # Capture 1 kS @ 100 kS / s on CH1.
    plt.plot(x, y)
```

This works for me on my V5. Do the curves overlap, or do they start all over the place? If the trigger is working, they should all start at 1 V or slightly above.

viric, 2022-11-17:
Ah sure, I tried a 50Hz signal, and the other was a serial line frame that came once in a while.

How can I get a normal trigger implementation? How does one compile a new firmware (on linux) and flash it to a v5 board? Only with a PIC programmer?

bessman, 2022-11-17:
The timeout can be increased, and even disabled entirely, with the `timeout` argument of `configure_trigger`. It's a prescaler, so a value of 1 doubles the timeout to 12.5 ms, 2 doubles it again 25 ms, and so on.

To disable the trigger timeout, set the timeout prescaler such that:
`timegap * 4 / timeout < 1`

The v5 has no bootloader, so flashing new firmware requires a hardware programmer like a PICkit, yes. If you can get ahold of one you can use it to flash the [bootloader](https://github.com/fossasia/pslab-bootloader) , after which new firmware can be flashed over USB.

The pslab-bootloader and pslab-firmware repos have instructions for how to build and flash them. It would be great if you could try it out, and let us know if anything is unclear or doesn't work as expected. 

Note that the firmware is currently undergoing a major overhaul. It's almost done and should be mostly feature complete, but there may be bugs we haven't ironed out yet. It's also not very well tested on the v5, but it should work in theory.

bessman, 2022-11-17:
On second thought, disabling the timeout is only possible for high sample rates / small timegaps < 4 us. This is because the timegap prescaler is only one nibble wide, i.e. its maximum value is 15.

Nivriti-ctrl, 2025-04-04:
Possible Software Fix for PSLab Trigger Issue (Issue #212)
I took a closer look at the configure_trigger() function in oscilloscope.py and noticed that it may be timing out too quickly (around 6.25 ms by default). This could be why low-frequency signals (like 50 Hz) cause the trigger to timeout before the condition is met. I believe that increasing the timeout—by using a higher prescaler value—might solve the issue.
def configure_trigger(
    self,
    channel: str = None,
    voltage: float = 0,
    prescaler: int = 0,
    enable: bool = True,
):
    if enable is False:
        self._trigger_enabled = False
        return

    if channel is not None:
        self._trigger_channel = channel

    # Determine the trigger channel index based on the channel name
    if self.trigger_channel == self._channel_one_map:
        ch_idx = 0
    elif self.trigger_channel in self._CH234:
        ch_idx = self._CH234.index(self.trigger_channel) + 1
    else:
        raise TypeError(f"Cannot trigger on {self.trigger_channel}.")

    # If no prescaler is provided, default to a higher value to extend the timeout
    if prescaler == 0:
        prescaler = 15  # This increases the timeout period

    self._device.send_byte(CP.ADC)
    self._device.send_byte(CP.CONFIGURE_TRIGGER)
    # Configure trigger: high nibble for timeout prescaler, low nibble for trigger channel mask
    self._device.send_byte((prescaler << 4) | (1 << ch_idx))
    level = self._channels[self.trigger_channel].unscale(voltage)
    self._device.send_int(level)
    self._device.get_ack()
    self._trigger_enabled = True

 I analyzed the PSLab Python trigger code and found that:
1️⃣ The configure_trigger() function may be timing out too quickly (6.25 ms).
2️⃣ Increasing timeout (timeout=15) should help, but it’s unclear if PSLab handles it dynamically.
3️⃣ The oscilloscope.py file does not allow setting a custom timeout easily.

 Suggested Fix:

Modify configure_trigger() to allow a higher default timeout (e.g., 25 ms instead of 6.25 ms).

Add an option to set timeout dynamically for low-frequency signals.

Would this be a good approach? Let me know if you need help modifying the code! 😊